ricardoquesada/esp-idf-arduino-bluepad32-template #1: Platformio, esp32, arduino framework

drmaus83, 2024-02-16:
Wondering if you could add instructions for using this with platformio using esp32 with arduino framework. I got it to work with arduino ide but recently switched to vs code and platformio as my skills have somewhat expanded to a point that i can us vs code and platformio. Idk if this is the correct place to ask this but it did say "feature requests" can go here and I dont have any of the socials you have posted and you dont have any email listed. The main issues is i dont have the skills/confidence to use ESPidf framework yet so itd be cool to use this in the environment i described above. Tia :)

ricardoquesada, 2024-02-16:
it should be possible.

Adding link to platform io documentation: https://docs.platformio.org/en/latest/frameworks/espidf.html#framework-espidf

drmaus83, 2024-02-16:
Ok, would you concider adding your email to your git profile? I would've sent a dm but I don't got the socials you have. Also I read those docs but it seems especially focused in the espidf frame work but I'm using the arduino frame work on platform io with my esp32 since I don't really understand espidf yet not sure I will for awhile since I've never really dealt with this type of stuff.

ricardoquesada, 2024-02-16:
fixed for esp-idf in `develop` branch.

not sure how to do it for non esp-idf... if you find any info, feel free to post it here