ricardoquesada/esp-idf-arduino-bluepad32-template #5: Cloned Main having compiling issues platformio vscode

GimpMaster, 2024-07-06:
Hello,

I just wanted to try out this on my esp32 using VSCode, Windows, PlatformIO, ESP-IDF. So I cloned main branch and then I tried to build and got this error.

```
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp-elf @ 13.2.0+20230928 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.41.0.windows.3") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:

[1/1] idf (4.4.7)

-- Project sdkconfig file C:/projects/redsky/bluepad32-test/sdkconfig.esp32dev
Loading defaults file C:/projects/redsky/bluepad32-test/sdkconfig.defaults...

-- Configuring incomplete, errors occurred!
See also "C:/projects/redsky/bluepad32-test/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Warning at C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Toolchain
  C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
  version 13.2.0 is not the supported version 8.4.0.  Check Getting Started
  documentation or proceed at own risk.

Call Stack (most recent call first):
  C:/Users/gimp3/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:6 (gcc_version_check)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/build.cmake:334 (include)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/build.cmake:543 (__build_process_project_includes)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:7 (project)

CMake Error at C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:37 (message):

  Toolchain:
  C:/Users/gimp3/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe,
  crosstool-ng version esp-13.2.0_20230928 doesn't match supported version
  esp-2021r2-patch5

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.

Call Stack (most recent call first):
  C:/Users/gimp3/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:7 (crosstool_version_check)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/build.cmake:334 (include)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/build.cmake:543 (__build_process_project_includes)
  C:/Users/gimp3/.platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:7 (project)

============================================================================================================================================= [FAILED] Took 3.93 seconds =============================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:03.925
======================================================================================================================================= 1 failed, 0 succeeded in 00:00:03.925 ======================================================================================================================================= 
```

The only suggestion in the output is to try to run `idf.py fullcelan`. THis does not seem to be in the registered path for the powershell terminal. I did noticed platformio has a build clean and I tried that but still failed with same error. Any suggestions?

Thanks!

GimpMaster, 2024-07-06:
Just a heads up I was able to solve this by specifying a certain platform for platform.io that is back on esp-idf version 4.X

I did this by adjusting platform.io to have the following

platform = espressif32@5.4.0

This forces it to be on 4.X of ESP-IDF instead of 5.X which is currently not compatible with Arudino ESP32 that this template is currently using.

ricardoquesada, 2024-07-08:
good to know.

Closing it since it was also discussed here:
https://discord.com/channels/775177861665521725/1259267610516656259/1259888833416138762

closed 2024-07-08 by ricardoquesada

GimpMaster, 2024-07-08:
@ricardoquesada - It would be great to update your template to change the platform.io to specify the platform = espressif32@5.4.0

Thank you for the awesome library!

ricardoquesada, 2024-07-08:
done. fixed. ty.