ricardoquesada/esp-idf-arduino-bluepad32-template #9: Using Wifi:`IRAM0 segment data does not fit.` Am I doing something wrong?

joaopedrotaveira, 2024-08-10:
Hi,

I use the template to create a program to control a pan-tilt assembly, controlled with gamepads. I also would like to add some REST API to it, but by simply adding `Wifi.begin` I got the error bellow. I reach the limit space to the all program or is this something else?

Thanks

```
Linking .pio/build/wemos_d1_uno32/firmware.elf
/Users/smiff/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/wemos_d1_uno32/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/smiff/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/Users/smiff/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 13000 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/wemos_d1_uno32/firmware.elf] Error 1
```

ricardoquesada, 2024-08-11:
Try with this doc:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html

joaopedrotaveira, 2024-08-12:
Thanks. Problem solved, ... for now.

I will need to shink something in the future, otherwise there's no left space for me.

```
Linking .pio/build/wemos_d1_uno32/firmware.elf
Retrieving maximum program size .pio/build/wemos_d1_uno32/firmware.elf
Checking size .pio/build/wemos_d1_uno32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  32.1% (used 105264 bytes from 327680 bytes)
Flash: [==========]  96.6% (used 1012669 bytes from 1048576 bytes)
Building .pio/build/wemos_d1_uno32/firmware.bin
esptool.py v4.4
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
================================= [SUCCESS] Took 37.71 seconds =================================
```

Thanks once again.

ricardoquesada, 2024-08-12:
no problem.

closed 2024-08-12 by ricardoquesada