ricardoquesada/esp-idf-arduino-bluepad32-template #10: Error compiling using ESPIDF + Platform IO

ste2425, 2024-11-25:
So I'm following the setup guide for Platform IO within VSCode.

I'm running this on Linux Mint 22 Cinnamon.

During Platform IO extension setup and when trying to compile i get the following error in the console.

```
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

ModuleNotFoundError: No module named 'pkg_resources':
```

It appears to be a bug in the version of Espresif32 platform used for the PlatformIO config according to this forum post: https://community.platformio.org/t/error-creating-projects-pkg-resources-issue/38481/2

This PR change fixes the issue: https://github.com/platformio/platform-espressif32/commit/bd7dd2cd602f895d4a21d4360933384139e6132f

Locally i manually applied that change and it now all compiles fine.

Just raising for awareness.

ricardoquesada, 2025-01-03:
Thanks. Did it got merge ?

There's has been some "politics" involved between PlatformIO and Espressif.
See:  https://github.com/platformio/platform-espressif32/issues/1225

in any case I updated the `platformio.ini` files both for `examples/esp32` and the "template app" , and everything seems to be working Ok when using ESP-IDF v5.3.

If you still have issues with PlatformIO, please LMK. ty

ricardoquesada, 2025-04-10:
the way `platform.ini` was setup, was to use latest esp-idf toolchain.

right now it will use arduino core 3.2.x + esp-idf 5.4.1

closed 2025-04-10 by ricardoquesada