ricardoquesada/esp-idf-arduino-bluepad32-template #11: How to access internal data

dragoncoder047, 2024-12-27:
I have a ps5 controller... but it seems there is no way to access the features specific to the ps5 controller via the Arduino api. For example, setting adaptive trigger effects, changing the player LEDs to something other than the 0-1-2-3-4 sequence the library assumes, and accessing the absolute position (not relative) of the fingers on the touchpad.

How can I access the backend data from the bluepad32 objects and operate on them directly?

I am using the "Arduino Core for ESP32 + Bluepad32" board package (board package version 4.1.0, IDE version 2.3.4).

ricardoquesada, 2024-12-28:
you might need to include `uni_hid_parser_ds5.h`:

https://github.com/ricardoquesada/bluepad32/blob/main/src/components/bluepad32/include/parser/uni_hid_parser_ds5.h

and then call the functions that are declared there...but call them from the "BTstack" thread:

https://bluepad32.readthedocs.io/en/latest/programmers_guide_raw/#multithreading

dragoncoder047, 2024-12-28:
>you might need to include `uni_hid_parser_ds5.h`

No, I figured I would have to include the bluepad32 guts or something...

But the Arduino BP32 api gives me a `Controller*`. How do I get the `struct uni_hid_device_s*`  is what I am asking.

dragoncoder047, 2024-12-28:
Also, it seems that the `ds5_send_output_report` and `ds5_output_report_t` are private (`static`) to the .c version of the file you linked -- but I would need to access those. Is there a way to use those functions that wouldn't involve forking this library and adding the declarations to the .h file myself, then somehow pointing the Arduino IDE to my fork instead?

ricardoquesada, 2025-01-01:
yeah... I'll need to make some changes... I guess I can add a function that returns the `uni_hid_device_s*`.

You should also know that Arduino code runs in its own task (different than BP32), so you will need to fix that. See:

https://bluepad32.readthedocs.io/en/latest/programmers_guide_raw/#multithreading

dragoncoder047, 2025-01-01:
> yeah... I'll need to make some changes... I guess I can add a function that returns the `uni_hid_device_s*`.

My suggestion would be something like `.c_dev()` or something like that, akin to the `String`'s `.c_str()` that returns the underlying `char*` array.

> You should also know that Arduino code runs in its own task (different than BP32), so you will need to fix that. See:
> 
> https://bluepad32.readthedocs.io/en/latest/programmers_guide_raw/#multithreading

Yeah, I know about that part... not a problem

ricardoquesada, 2025-01-01:
Added `arduino_get_internal_hid_device()`.

Please LMK if that works for you:

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/commit/c3709b365123cf025b34b6a0054db63a74a1a44a

ricardoquesada, 2025-01-01:
I updated the docstring here:

https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/components/bluepad32_arduino/include/arduino_platform.h#L90

dragoncoder047, 2025-01-02:
> Please LMK if that works for you

Looks like it should, except I can't figure out how to install it from git. I did:

```sh
cd ~/Documents/Arduino
mkdir -p hardware/bluepad32 && cd hardware/bluepad32
git clone https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template.git esp
cd esp
git checkout develop
```
and then restarted the Arduino IDE, but the new core doesn't show up. Can you tell what I am doing wrong?

ricardoquesada, 2025-01-02:
ah... yes, this is the template project. It won't work from Arduino IDE.

The  Bluepad32 Core (the one that can be used from Arduino IDE) is not ready yet.

The "template project" uses Arduino Core, but cannot be used from Arduino IDE.

Arduino IDE board will be updated soon.

dragoncoder047, 2025-01-02:
> The Bluepad32 Core (the one that can be used from Arduino IDE) is not ready yet.

So you're saying the core that I installed and used initially was a *pre-release*? Wow!!

ricardoquesada, 2025-01-02:
no, not a pre-release. I guess you are using Bluepad32 Arduino Core v4.1.0.

I fixed it on "develop" branch in the template project. This will be part of v4.2.0

First I need to release:
  - Bluepad32 v4.2.0
  - Then I'll release Bluepad32 template project v4.2.0 (This is for Arduino users that don't use Arduino IDE)
  - After then I'll release the Arduino "board" v4.2.0 (which will be based on the template v4.2.0)

dragoncoder047, 2025-01-02:
ok, lmk when you have done that, no rush.