ricardoquesada/esp-idf-arduino-bluepad32-template #13: platformio.ini again

Grey-Lancaster, 2025-02-22:
I actually built it yesterday on my old computer and it worked fine.  Then I got a new computer, Installed VS and Platformio cloned the repo and it gave an idf version mismatch error.

I changed from develop to main and it works. I see develop had an update yesterday

`platform = https://github.com/pioarduino/platform-espressif32.git#main`

PS: my old laptop took 55 mins to build.  My "new" refurbished lappy only took 6 mins. That's good enough for me :-)

Modac, 2025-04-01:
For me this worked: 
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13+github/platform-espressif32.zip`

szelenka, 2025-04-02:
just ran into this when moving PC also, I'm not that knowledgable on pioarduino, but it seems pinning to main/develop is going to always pull in the latest change; which is 5.4.2 in this case.

```ini
[platformio]
src_dir = main

[env]

platform = https://github.com/pioarduino/platform-espressif32.git#53.03.13
```

Do we know what dependency is preventing this example from working with later versions of the ESP IDF?

Is it only the component/arduino?
https://github.com/espressif/arduino-esp32/releases

ricardoquesada, 2025-04-09:
please see:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/issues/14

I'll update Arduino Core to v3.2.0, which uses ESP-IDF v5.4.

ricardoquesada, 2025-04-10:
fixed in `main` branch.

closed 2025-04-10 by ricardoquesada