ricardoquesada/esp-idf-arduino-bluepad32-template #14: Getting a depends error when using PlatformIO

baget, 2025-04-07:
Clone the repro according to the guide but getting errors,
tried to delete ~/.platformio without any luck

Here is the Error:

> -- Configuring incomplete, errors occurred!
> 
> fatal: not a git repository (or any of the parent directories): .git
> CMake Error at C:/Users/user/.platformio/packages/framework-espidf/tools/cmake/build.cmake:610 (message):
>   ERROR: Because project depends on idf (>=5.3.0,<5.4.0) which doesn't match
>   any versions, version solving failed.
> 
> Call Stack (most recent call first):
>   C:/Users/user/.platformio/packages/framework-espidf/tools/cmake/project.cmake:717 (idf_build_process)
>   CMakeLists.txt:7 (project)

ricardoquesada, 2025-04-09:
what esp-idf version are you using ?
It requires, as the error says, v5.3.x

baget, 2025-04-09:
right, I didn't install ESP-IDF, it download automatically 
After some debugging, If I upgrade the Arduino Folder to the latest, it works.
but I need to manually update it

ricardoquesada, 2025-04-09:
ahhh... arduino core 3.2 was released a few weeks ago, and it is based on esp-idf 5.4... I'll update it later tonight

ricardoquesada, 2025-04-10:
fixed in `main` branch.

v4.2.1

closed 2025-04-10 by ricardoquesada